agnunez/espros #3: Can't connect to ROS master

Loreno10, 2017-12-04:
Hello,
I've just tried using he library to ocnnect to ROS and I can't for some reason.

Here's my Arduino sketch tha creates a simple publisher:
`#include <ESP8266WiFi.h>
#include <ros.h>
#include <std_msgs/String.h>
#include <std_msgs/Int16.h>
#include <std_msgs/Float64.h>

//WIFI
const char* ssid = "Redmi 4X";
const char* password = "homologacja";

IPAddress server(192,168,172,129);
IPAddress ip_address;
WiFiClient client;

int status = WL_IDLE_STATUS;

class WiFiHardware {

  public:
  WiFiHardware() {};

  void init() {
    // do your initialization here. this probably includes TCP server/client setup
    client.connect(server, 11411);
  }

  // read a byte from the serial port. -1 = failure
  int read() {
    // implement this method so that it reads a byte from the TCP connection and returns it
    //  you may return -1 is there is an error; for example if the TCP connection is not open
    return client.read();         //will return -1 when it will works
  }

  // write data to the connection to ROS
  void write(uint8_t* data, int length) {
    // implement this so that it takes the arguments and writes or prints them to the TCP connection
    for(int i=0; i<length; i++)
      client.write(data[i]);
  }

  // returns milliseconds since start of program
  unsigned long time() {
     return millis(); // easy; did this one for you
  }
};

void setupWiFi()
{
  WiFi.begin(ssid, password);
  Serial.print("\nConnecting to "); Serial.println(ssid);
  uint8_t i = 0;
  while (WiFi.status() != WL_CONNECTED && i++ < 1000) delay(500);
  
  if(i == 1001){
    Serial.print("Could not connect to"); Serial.println(ssid);
  //  while(1) delay(500);
  }
  
  if(WiFi.status() == WL_CONNECTED)
  {
    Serial.print("Ready! Use ");
  Serial.print(WiFi.localIP());
  Serial.println(" to access client");

  }
}

ros::NodeHandle_<WiFiHardware> nh;

std_msgs::String str_msg;
ros::Publisher chatter("chatter", &str_msg);

char hello[13] = "hello world!";

void setup()
{
  Serial.begin(115200);
  setupWiFi();
  delay(2000);

  
  nh.initNode();
  nh.advertise(chatter);
}

void loop()
{
  str_msg.data = hello;
  chatter.publish( &str_msg );
  nh.spinOnce();
  delay(1000);
}`

On Ubuntu, I first run _roscore_. Then I do `rosrun rosserial_python serial_node.py tcp`.
All I see in the Terminal is this:
`[INFO] [1512416583.799159]: ROS Serial Python Node
Fork_server is:  False
[INFO] [1512416583.816673]: Waiting for socket connections on port 11411
waiting for socket connection
`

After that I turn on my ESP8266. From the Serial Monitor I see that it successfully connects to WiFi. Unfortunately Ubuntu Terminal does not inform about connection. I also cannot see my topic in ROS.
Do you have any idea what might be the issue?

Both Ubuntu and ESP8266 are connected to the same WiFi network of course.

mosu7, 2018-06-18:
Hi, we've met the same problem. Did you solve it?

Loreno10, 2018-06-18:
I downgraded ROS on the server to kinetic I think. To be exact, I downgraded rosserial only and it worked.

Loreno10, 2018-06-18:
No problem, update this thread later with your result

mosu7, 2018-06-18:
Hi, I downgraded it and tried your code.  It showed this.
`waiting for socket connection
[INFO] [1529351133.333148]: Established a socket connection from 192.168.1.112 on port 49157
[INFO] [1529351133.334003]: calling startSerialClient
[INFO] [1529351135.436264]: Requesting topics...
[INFO] [1529351135.438529]: RuntimeError exception caught
[INFO] [1529351135.439295]: startSerialClient() exited`

Yugal2531, 2020-01-26:
@Loreno10 whats the rosserial version you have used

Loreno10, 2020-01-26:
> @Loreno10 whats the rosserial version you have used

Oh man, it was quite long ago that I had this issue, I don’t remember. Nowadays I don’t have to use ros anymore and I can live my life happily.

Yugal2531, 2020-01-28:
> > @Loreno10 whats the rosserial version you have used
> 
> Oh man, it was quite long ago that I had this issue, I don’t remember. Nowadays I don’t have to use ros anymore and I can live my life happily.

What u use today😆?
Btw I got my esp8266 connected to ROS over WiFi, Successfully!!!😄

Yugal2531, 2020-01-28:
> > > @Loreno10 whats the rosserial version you have used
> > 
> > 
> > Oh man, it was quite long ago that I had this issue, I don’t remember. Nowadays I don’t have to use ros anymore and I can live my life happily.
> 
> What u use today?
> Btw I got my esp8266 connected to ROS over WiFi, Successfully!!!

Will push my code soon..

viso951, 2021-03-12:
@Loreno10 How have you solved it? I have the same problem, I am using ros noetic, Ubuntu 20 and the latest version of rosserial