agnunez/espros #4: Doesnt compile for esp32 (Nodemcu v2)

chrissunny94, 2018-01-12:


PawanRao93, 2018-11-06:
@chrissunny94 - did you find any solutions for the problem. I am using a NodeMCU - ESP-32S module and as you said, it does not compile and I get the same error. However, when I compile choosing the board to be a "Generic ESP8266 Module" the code compiles with no errors. However, I am unable to talk to the NodeMCU - ESP32 board with the board selected to Generic ESP8266 Module. 

Please let me know in case you found any workaround for this problem. 

@agnunez  - will be very helpful if you can help us with this problem. 

Thanks, 
Pawan Rao

akshaykvnit, 2019-07-08:
For ESP32, you need to change `#include <ESP8266WiFi.h>` to `#include <WiFi.h>`

closed 2020-05-03 by agnunez