agnunez/espros #7: NodeMCU ESP32 : Protocol version of client is unrecognized, expected Rev 1 (rosserial 0.5+)

sunnysetia93, 2019-02-14:
I have **ROS** **melodic** installed on my **Master Linux machin**e where I have my roscore running. 

I have serially connected **NodeMCU esp32** to the linux machine, where esp32 publishes data of an ultrasonic sensor on a topic which is then subscribed by node running on a linux machine.

I get this error when I am trying to run 
_rosrun rosserial_python serial_node.py /dev/ttyUSB0_ 
**Error**: Protocol version of client is unrecognized, expected Rev 1 (rosserial 0.5+)

that is why when i run rostopic echo /topicname, there is no such topic found.

Also, When I run the same code on **arduino UNO**, everything works as expected.

Kindly help me with a detailed explanation as I am a beginner in ROS and cannot proceed further.

sujoy743175, 2021-03-16:
hi @sunnysetia93  ... you must have solved the problem ... can you help me to come of the same ... m using  neotic and foxy . thanx...

DeepakKumarShah, 2021-09-17:
@sujoy743175  

https://github.com/ros-drivers/rosserial/issues/354   here some people seem have solved this problem. If you understand their there discussion then please explain me too, I am not good at programming.

DeepakKumarShah, 2021-09-17:
It solved my problem.

[image]

joedavid-tuni, 2023-08-04:
My ESP-32 was running at 240Mhz, when I changed it to 160Mhz this error was gone. You change it in the Arduino IDE tools>CPU frequency

AbrarMahmud, 2024-07-16:
> It solved my problem.
> 
> [image]

It worked for me too.---Thanks---